RoganDawes/P4wnP1 #341: No sessions when trying to gain shell access

SirQuacksALot, 2020-11-05:
When im attempting to gain shell access with p4wnp1 aloa alpha, I get no active sessions when trying to gain shell access. plz help

KatzeMau, 2020-11-25:
Open an issue without any details ....

ishan-saha, 2020-12-23:
i am facing the same issue and the error that i could see at most is as bellow 
root@kali:~/P4wnP1/dist/scripts# ./wifi_covert_channel.sh 
Kill old hidstager processes ...
Starting HID stager for WiFi covert channel payload
HID covert channel server already running
LED blink count set to 3
root@kali:~/P4wnP1/dist/scripts# Delivering " /usr/local/P4wnP1/legacy/wifi_agent.ps1 " via raw HID device " Error: "
Exit after first delivery
Traceback (most recent call last):
  File "/usr/local/P4wnP1/legacy/hidstager.py", line 169, in <module>
    main(sys.argv[1:])
  File "/usr/local/P4wnP1/legacy/hidstager.py", line 165, in main
    deliverStage2(outputfile, stage2, oneshot)	
  File "/usr/local/P4wnP1/legacy/hidstager.py", line 96, in deliverStage2
    with open(hidDevPath,"r+b") as f:
IOError: [Errno 2] No such file or directory: 'Error:'

ishan-saha, 2020-12-23:
I am facing the same issue and couldn't narrow it down but after a bit of searching i could:
`root@kali:~/P4wnP1/dist/scripts# ./wifi_covert_channel.sh 
Kill old hidstager processes ...
Starting HID stager for WiFi covert channel payload
HID covert channel server already running
LED blink count set to 3
root@kali:~/P4wnP1/dist/scripts# Delivering " /usr/local/P4wnP1/legacy/wifi_agent.ps1 " via raw HID device " Error: "
Exit after first delivery
Traceback (most recent call last):
  File "/usr/local/P4wnP1/legacy/hidstager.py", line 169, in <module>
    main(sys.argv[1:])
  File "/usr/local/P4wnP1/legacy/hidstager.py", line 165, in main
    deliverStage2(outputfile, stage2, oneshot)	
  File "/usr/local/P4wnP1/legacy/hidstager.py", line 96, in deliverStage2
    with open(hidDevPath,"r+b") as f:
IOError: [Errno 2] No such file or directory: 'Error:'

`